LakeMaps/boat #1: Maestro Communication Interface

arandell93, 2016-07-26:
The [Pololu Maestro](https://www.pololu.com/docs/0J40) has a couple of options for communication.

There is the [hardware level](https://www.pololu.com/docs/0J40/5.a) interface which determines whether you use USB or UART and the settings of the selected port.

Then there is the [command protocol](https://www.pololu.com/docs/0J40/5.c) which determines the structure of the message being sent over the serial line.

These need to be configured on the Maestro itself using the [Maestro Control Center](https://www.pololu.com/docs/0J40/4) prior to connecting to the control system.

TL;DR: How will we talk to the Maestro? My vote is USB + Pololu Protocol for familiarity and reliability of USB.
However, we could use the UART and make the Maestro live on a daughterboard connected to the Pi.

whymarrh, 2016-07-29:
USB + Pololu Protocol sounds like a reasonable default, yup. (Note that the command protocol, IIRC, assumes a single Maestro connected.)

> [W]e could use the UART and make the Maestro live on a daughterboard connected to the Pi.

What would the benefits of this setup be?

arandell93, 2016-07-29:
> What would the benefits of this setup be?

Consistency across device interface, more of an electrical thing than a software thing.

whymarrh, 2016-07-30:
Where both USB and UART are forms of serial (?) I can't imagine this making much of a difference in software. We'd still be speaking the Pololu Protocol, yeah?

If there are any upsides electrical design-wise, then let's use that.

arandell93, 2016-08-02:
We will still be speaking Pololu Protocol, yeah.
There are pros and cons electrically. I'm not sure what I prefer to be honest. I think I like USB better because it's a little less awkward than interfacing a maestro with a PCB.

whymarrh, 2016-10-26:
I think this has been worked out—we won't be using a Maestro for the time being.

closed 2016-10-26 by whymarrh