LakeMaps/boat #2: Investigate Using GPIO as UART on Raspberry Pi

arandell93, 2016-07-26:
The RasPi has a single UART port on its [GPIO](https://en.wikipedia.org/wiki/Raspberry_Pi#General_purpose_input-output_.28GPIO.29_connector). 
We need to use some of the generic GPIO as serial ports, which can be acheived through 'bit-banging' the GPIO.
Here's a look at the GPIO configuration:
[image]
Ideally, we would want as many devices as possible. Let's say 8 plus the hardware UART for a total of 9.
This means we need 16 GPIO pins in order to create 8 UART ports.
Avoiding the I2C and SPI pins (in case we want to use them down the road), we can configure the following pins (pairing is not important):
17, 27, 22, 10, 05, 06, 13, 19, 26, 18, 23, 24, 25, 12, 16, 20, 21
That's a total of 17 pins, and we only need 16.

Here are a few quick links I found relating to configuring the GPIO to act as UART ports:
[[1]](http://www.rs-online.com/designspark/electronics/eng/blog/raspberry-pi-2nd-uart-a-k-a-bit-banging-a-k-a-software-serial) [[2]](http://raspberrypi.stackexchange.com/questions/27488/pigpio-library-example-for-bit-banging-a-uart)

whymarrh, 2016-07-31:
How many proverbial eggs have we put into the "bit-banging" with GPIO basket?

In reading through some materials related to the process, it looks like it requires a real-time OS, which Linux is not.<sup>\[citation needed\]</sup> (At least by default—apparently there's an RT patch, the `PREEMPT_RT` patch, but I don't really think it's something I want to get into if we can avoid it.) Additionally, even if we had a RTOS, bit-banging as a concept seems to me, as of right now, fundamentally flaws and super prone to error.<sup>[\[1\]](http://raspberrypi.stackexchange.com/a/8958) [\[2\]](http://raspberrypi.stackexchange.com/a/1991)</sup> I'm not sure that I want to design anything around this hack.

arandell93, 2016-08-01:
I appreciate that it's a bit of a hacked solution in terms of software. I did a bit more digging and came across [this](https://www.raspberrypi.org/forums/viewtopic.php?f=9&t=2006) forum post which discusses using an I2C or SPI chip to create a UART, as well as some USB options. It's worth discussing those as potential solutions. 

The [MAX3109](https://www.maximintegrated.com/en/products/interface/controllers-expanders/MAX3109.html) contains a dual UART TX/RX with an SPI/I2C interface. We send data via our preferred interface and it sorts out the 'make the volts' part of UART following all require rules. The thing I'm not sure of is how we configure the setting on the chip for things like baud, parity, # data bits, etc. We would need 4 of these chips to give us 8 UART.

A USB to quad UART chip such as the [FT4232H](http://www.ftdichip.com/Products/ICs/FT4232H.htm) seems to be relatively easy to work with as well. We could have two of those to max 8 UART, but then we're starting to get a mess of USB cables hanging out of the Pi, and wires take up space.
There's also the the option of a chip per UART, using something like the [MCP2200.
](http://www.microchip.com/wwwproducts/en/MCP2200) However, we'd need 8 of these on one board and some USB complexity arises.

We're gonna need input from @FifoIronton for this one as it impacts the design of Dartmouth significantly. Any chips we select can't be [SMD](https://en.wikipedia.org/wiki/Surface-mount_technology) as we can't solder to them. We also risk making Dartmouth pretty big if we need a lot of chips. If appropriate, we can move the hardware side of the discussion over to the [Dartmouth](https://github.com/LakeMaps/Dartmouth) repo but I think at least part of the discussion belongs here as there are software considerations related to the interface type.

arandell93, 2016-08-03:
Closing.
We have selected USB-UART as the hardware interface.
Discussion of hardware selection and integration is moving to [Dartmouth](https://github.com/LakeMaps/Dartmouth) repository ([Issue #5](https://github.com/LakeMaps/Dartmouth/issues/5)).

arandell93, 2016-08-09:
@whymarrh: Myself and @FifoIronton have done a sufficient amount of research and are confident in saying that the 'software serial' method is the best fit for us at such low baud rates (4800 b/s).
This saves us $30-50 and huge design and assembly time on the Dartmouth board so the design choice is justified.
We'll go ahead and pursue bit-banging the GPIO as the method by which we obtain 8 extra UART, all at 4800 baud. @FifoIronton and yourself can decide the pin assignments.
We will reserve the hardware UART so that in the event certain devices require higher baud we can still implement them.

arandell93, 2016-11-08:
No longer required as we are using the 1 Pi + N Arduino design and each Arduino will connect via USB.

closed 2016-11-08 by arandell93