LakeMaps/boat #3: Wireless Link Architecture

arandell93, 2016-07-26:
### Hardware
For the wireless link, it looks like we'll be using some low bandwidth, serial radios for the time being.
Most likely something like and [XBee Pro 900HP](http://www.digi.com/products/xbee-rf-solutions/modules/xbee-pro-900hp).

### Software
We will need software on both the boat and on the shore to make this system work.
Presumably, we would have a laptop on the shore with the XBee plugged in acting as a serial port for the computer. Similarly, the XBee would be a serial port on the RasPi.

### Message Structure
We need to develop a packet structure for shore-boat and one for boat-shore.
I propose that the boat initiates the message (since it is more important we receive information from the boat than the boat receive it from us in the event of a problem).
I'm not sure what already exists, and certainly we don't need to reinvent the wheel if something will suit our needs. If nothing exists, I propose the following as a starting point:

Boat Packet:
1. New Message Indication (Byte)
2. Control Mode Status (auto/remote/return home/path upload/standby) (byte)
3. Current Coordinates (Lat/Long) (pair of floats)
4. Current Depth below Transducer (in decimal meters) (float)
5. Range to Next Waypoint (in meters) (word)
6. Heading to Next Waypoint (in degrees) (word)
7. Current Heading (in decimal degrees) (float) 
8. Current Speed (in decimal meters/second) (float. Could be a byte that we scale by 10 ie: 255 becomes 25.5)
9. Wireless Signal Strength (in dB) (byte)
10. Battery Voltage (in decimal Volts) (float. Could be a byte that we scale by 10 ie: 255 becomes 25.5)
11. Power Consumption (in decimal Watts) (float)
12. Sequential message number/counter (word)
13. Checksum

Shore Packet:
1. New Message Indication
2. Control Mode Command (auto/remote/return home/path upload/standby) (byte)
3. Throttle (For Remote Control, -1 to 1 representing full reverse to full forward)
4. Steering (For Remote Control, -1 to 1 representing all the way left to all the way right)
5. Path Upload (Number of new waypoints incoming, with the MSB reserved for true=override current path and false=add on to end of path) (word)
6. New Path Data (if no new path being uploaded, send a pair of null floats) (pairs of floats)
7. Sequential message number/counter (word)
8. Checksum

FifoIronton, 2016-07-27:
Hey, I was just reviewing some stuff, and it looks like we may have (unfortunately) gotten confused. The RFM69 is not a LoRa based module, but just a regular packet radio solution. 

The LoRa radio module that is used on the boards is actually the [RMF95W](http://www.digikey.ca/product-detail/en/rf-solutions/RFM95W-868S2/RFM95W-868S2-ND/5051755), which costs a lot more ($32), but has a maximum sensitivity of -148dB, with the same output power according to the data sheet.

It depends on how much we're willing to pay, our calculations from last night made the RFM69 look promising, but the LoRa ones are definitely a little better.

Something to think about.

arandell93, 2016-07-27:
Hmmmmm... That makes the choice harder. the extra sensitivity is definitely a plus, but at four times the cost it's hard to decide.
I wonder what difficulty we're going to have configuring these chips to do what we want.

In any case, their footprint and pinout seem to be the same, so we could start off with the cheap ones and if we find we need the extra strength we can get the LoRa modules and just put them on our existing daughter boards. Let's design the modules to be removable from the daughter board and proceed as we were planning with the RFM69.

whymarrh, 2016-10-21:
refs LakeMaps/hangashore#3

FifoIronton, 2016-10-30:
Looking around at how other people use the RFM modules, there are a couple Arduino libraries that exist that we could at least poke through to see how they work. Guides for testing the modules with these libraries exist [here](https://learn.adafruit.com/adafruit-rfm69hcw-and-rfm96-rfm95-rfm98-lora-packet-padio-breakouts/rfm69-test), with info on how to connect the pins and everything, which we should look into on Monday.

arandell93, 2016-11-02:
Note: Maximum packet size of the RFM69HCW is 61 bytes.

arandell93, 2016-11-08:
Since there is now an Arduino on each end of the wireless link, the scope of this issue has changed from the perspective of the boat control software.
We now have an interface between the Pi and an Arduino (UART).

arandell93, 2016-12-16:
Closing as we have established a wireless link via the [Wireless Module](https://github.com/LakeMaps/microcontrollers/wiki/Wireless-Communication-Microcontroller-API).

closed 2016-12-16 by arandell93