LakeMaps/boat #4: Define Control Modes

arandell93, 2016-07-27:
The boat will need several control modes that we can switch between (via commands on the wireless link) for different circumstances. Discussion can be had here about what control modes to have and what they should be used for.

I'll edit this post as we come up with more.
1. Autonomous - Boat autonomously navigates to a set of waypoints in order
2. Standby - Boat waits for further instructions/change in control mode
3. Remote - Boat receives throttle and steering commands over the wireless link
4. Return Home - Boat clear the waypoint queue and replaces it with a single location, the launch point
5. Path Upload - Similar to standby, but here the boat is listening for a new series of waypoints to be sent over the wireless link (Can we just use standby for this)
6. Set Home - Standby, but the boat sets its current location as its new home location

whymarrh, 2017-02-03:
@arandell93 does this issue have a home in a particular revision?

arandell93, 2017-02-03:
Starting in [Rev 1.3](https://github.com/orgs/LakeMaps/projects/5) we will need to begin adding control modes in order to switch between our basic 'autonomous' mode and RC. The modes we will need for Rev 1.3 will be:

1. Remote Control Mode (The mode we currently have)
2. Single Waypoint Autonomous Mode (as described in Rev 1.3)
3. Return Home Mode (requires ability to 'set' the home location via Hangashore and have it transmitted and stored on the boat)

whymarrh, 2017-03-16:
Should we move this to the wiki?

arandell93, 2017-03-16:
We should, but perhaps not until we have 1.3 implemented so that we have control modes.

whymarrh, 2017-05-04:
> Starting in [Rev 1.3][1] we will need to begin adding control modes in order to switch between our basic 'autonomous' mode and RC. The modes we will need for Rev 1.3 will be:
>
> 1. Remote Control Mode (The mode we currently have)
> 2. Single Waypoint Autonomous Mode (as described in Rev 1.3)
> 3. Return Home Mode (requires ability to 'set' the home location via Hangashore and have it transmitted and stored on the boat)

Do we want to elaborate on these modes:

- How are modes toggled?
- What happens when the boat transitions between the modes?

  [1]:https://github.com/orgs/LakeMaps/projects/5

arandell93, 2017-05-10:
> How are modes toggled?

A joystick button and/or soft button on the GUI would be the most obvious choices IMO. However, we should put in a caveat for some sort of 'heartbeat' such that if wireless comms is lost for a period of time the boat will automatically transition to 'return home' and begin moving toward the last known shoreside position.

> What happens when the boat transitions between the modes?

Following a request for a mode change, the boat will need to clear all outputs (set motor speeds to zero) and then do some form of acknowledgement to us that it is indeed transitioning (could be a message over wireless to indicate which state is is transferring from and to which state is it moving). Once the boat has confirmed that transition is complete, we then can begin giving it commands*

*The exception to this is that if the transition is to the 'return home' mode, the boat should simply tell us it has completed the mode transition and then begin moving home. That way if the link it lost it will still return home.