LakeMaps/boat #7: Define Scope of Shoreside Application

arandell93, 2016-07-29:
We require a computer-based application in order to facilitate the wireless link and control of the boat under various circumstances. This issue will be used to define the **scope** of the application but not the specific design details.

### Requirements
- Interface with wireless link (via USB port)
- Parse messages received from boat (#3)
- Present information from boat as appropriate in an intuitive manner (GUI)
- Interface with an input device (keyboard or, more appropriately, a gamepad)
- Interpret coordinate information from boat and interface with mapping software. If we re-assemble the GPS and Sonar information in NMEA sentences and put it on a virtual serial port this is easily possible. We can even do real-time tracking/map generation in [Reefmaster](http://reefmaster.com.au/index.php/products/reefmaster).
- Gather new path information from mapping software for export (Similar to [Ardupilot Mission Planner](http://ardupilot.org/planner/docs/common-planning-a-mission-with-waypoints-and-events.html))
- Assemble a message for transmission to the boat based on inputs, new paths (#3)
- Save received information to a local database
- Provide visualizations for logged data (other than coordinates on a map) such as speed, depth,

whymarrh, 2016-07-29:
We can leave this in here for now, but this project will likely end up in its own repository.

arandell93, 2016-07-29:
That's a good idea actually. We can reference between repositories right?
Let's come up with a clever name for the shoreside application and then make the repo.
Suggestions:
- LandLubber
- Hangashore (hang-a-shore)
- Lighthouse
- Foghorn
- Marconi

whymarrh, 2016-07-30:
> Hangashore

This seems to have a lot of meanings: [**angishore**](http://www.heritage.nf.ca/dictionary/#52)

arandell93, 2016-08-03:
Closing this and opening an issue in the [hangashore](https://github.com/LakeMaps/hangashore/issues) repo.
I'll add more specifics to the requirements when I do so.

closed 2016-08-03 by arandell93