LakeMaps/boat #10: Develop Differential Drive Motion Control Scheme

arandell93, 2016-10-26:
The boat is going to be controlled using differential drive. That is, we have two motors that will be used to propel and steer the boat. There will be no rudder on the boat.
[image]

In order to achieve this method of control via remote control, we have two options:
1. Differential Input
2. Throttle/steer input

### Differential Input###
This is the simpler of the two input control methods, but it is less intuitive to the user. In this method we use two separate input axis; one axis controls each motor. This would be similar to the way tanks, excavators and other similar tracked vehicles are controlled. The control inputs would be:
- Port Forward
- Port Reverse
- Stbd Forward
- Stbd Reverse

I recommend that if this is the control method chosen we apply the following development process in order to decrease initial development time:
- Keyboard or other digital input device that uses four buttons, two per 'axis' of control. This would be a binary control (on/off) over each direction.
- Gamepad with dual analog joysticks that allows for a scaled power to each axis. This would give the ability to steer with various rates of turn which would make the vehicle maneuver better.

If the development of an analog control scheme would not in reality impact the time required to develop then it would be prudent to skip the binary control in favour of analog control.

### Throttle/Steer Input ###
As opposed to a differential drive where the user is directly controlling the output of each motor, a throttle/steer input is more intuitive from the users perspective, as this is how control is typically done on things such as RC vehicles, video games, and real world cars.

The control inputs here are throttle (full reverse to full forward) and steering (full left to full right). Some math is required to translate these into outputs to the motors to propel the boat in the desired way. @cgregory52 should be able to come up with an intelligent scheme for controlling in this way, but it will definitely need to be tweaked depending on the real-world performance of the boat.

I recommend that if this is the control method chosen we apply the following development process in order to decrease initial development time:
- Keyboard or other digital input (Similar to a WASD setup) where the control is provided as below. This eliminates the math required for an analog implementation while still providing more intuitive control than differential inputs.
 - One key for full forward (Port 100%, Stbd 100%)
 - One key for full reverse (Port -100%, Stbd -100%)
 - One key for full turn to port (Port -100%, Stbd 100%)
 - One key for full turn to stbd (Port 100%, Stbd -100%)
-  Gamepad with dual analog joysticks that allows for a scaled power to throttle and steering, giving full maneuverability.

For this control scheme, the difference in time required to implement step 2 vs step 1 is much more obvious so for the upcoming deadline we would target digital control only.

whymarrh, 2016-10-26:
> For this control scheme, the difference in time required to implement step 2 vs step 1 is much more obvious so for the upcoming deadline we would target digital control only.

Is "digital control" supposed to be "differential control"? Anyhow....

> [A] throttle/steer input is more intuitive from the users perspective, as this is how control is typically done on things such as RC vehicles, [vidya games], and real world cars.

Yes, we'll do this one.

> Gamepad with dual analog joysticks that allows for a scaled power to throttle and steering, giving full maneuverability.

We'll do something like so:

[image]

arandell93, 2016-10-26:
> Is "digital control" supposed to be "differential control"?

Nope I meant digital control. As in buttons, like a D-pad.

> We'll do something like so:

So left analog stick for steering, R2 trigger for forward speed, L2 trigger for reverse speed? Make sure to handle the conflict of L2 and R2 both being pressed.

Also we totally need a horn now...

whymarrh, 2016-10-26:
Ah, ok. I think starting with analog control is fine—there isn't much of a difference in development time.

whymarrh, 2016-10-26:
Also, yes—left analog stick for steering, RT trigger for forward speed, and LT trigger for reverse speed.

cgregory52, 2016-10-26:
Request for added features: cruise mode. Press or hold a button to keep the last state without further input.

arandell93, 2016-10-26:
> Request for added features: cruise mode. Press or hold a button to keep the last state without further input.

I like it but I can see it causing problems in the event of a loss of communications or other problems.
As long as there's a loss-of-comms timeout that stops the motors it would be okay though.

arandell93, 2016-12-14:
We ended up implementing the system per Whymarrh's last comment for Rev 1.0

closed 2016-12-14 by arandell93