LakeMaps/boat #11: Reorganize Repository for Arduino + Pi Combination

arandell93, 2016-11-07:
Since we now have an Arduino onboard, I think the boat repo needs reorganization to reflect this.
I'm assigning this to @whymarrh since he knows how to github.

whymarrh, 2016-11-07:
What needs to change? This can continue to be a repository for boat control software.

arandell93, 2016-11-07:
I just figured since we have two platforms to develop for we would want to restructure.
I need to learn how to git I guess.

whymarrh, 2016-11-07:
Ah, no worries. This can remain as the repository for the boat control software (which will be independent/[decoupled](http://softwareengineering.stackexchange.com/a/244478/114526) from the Arduino sketches). Once we're ready we can create one or more repositories for each of the Arduino sketches that we will be using.

closed 2016-11-07 by whymarrh