LakeMaps/boat #12: (Rev 1) Pi-Arduino Communication protocol

arandell93, 2016-11-08:
Since we are using the 1 Pi + N Arduino system design, we need to develop a consistent communication protocol to implement across the board in order to communicate between the Pi and each sub-system.
This means that the wireless link #3 will also need to follow this protocol.

It is in our best interest to use the fastest serial speed possible for this (115200 bit/s) as it will take up less time on both ends to send/receive. We can standardize the payload size across all devices, or we can have each device send its own specifically sized payload. I'm not sure which I like more. Thoughts?

I propose something similar to the following. Feedback is welcome.
- Header
  - New Message Indicator [byte]
  - Sender ID [byte]
  - Recipient ID [byte]
  - Size of payload [byte]
- Payload
  - Size and contents vary with device
- Footer
  - Checksum [2 bytes]

whymarrh, 2016-11-09:
You might be interested in this talk: [How To Design A Good API and Why it Matters](https://www.youtube.com/watch?v=aAb7hSCtvGw)

whymarrh, 2016-11-09:
Another decent resource: [Designing Painless Protocols](https://nerdland.net/2009/12/designing-painless-protocols/)

From that:

> If your protocol uses multi-byte binary-encoded numbers, and it does not use network byte ordering, then you have committed a mortal sin. Network byte ordering is called what it is for a reason – so that machines communicating on a network have a common standard for which byte ordering to use when interchanging data with one another.

> [P]lease specify your protocol’s byte ordering in your documentation, even if it is little-endian for some (hopefully historical) reason. Otherwise, you will end up with people who write implementations that don’t do any conversion at all from their native ordering, and then then devices with different endianness can’t even talk over your protocol at all.

Also,

> 6\. Design for expansion
> 7. Don't be stingy with information
> 8. Document your protocol precisely

whymarrh, 2016-11-09:
For context, if the command to send data via the wireless link accepts an arbitrary payload, we would use something like [Protocol Buffers](https://developers.google.com/protocol-buffers/) to serialize information from Hangashore to the boat.

arandell93, 2016-12-14:
See [Microcontroller Wiki](https://github.com/LakeMaps/microcontrollers/wiki/Protocol-Syntax) for details on this API.

closed 2016-12-14 by arandell93