LakeMaps/boat #13: Develop Remote Access to Pi

arandell93, 2016-12-14:
In order to ease startup/teardown we need to be able to remotely access the Pi. Here are out options for access:

1. WiFi router onboard boat with wired link to Pi and wireless link to Shoreside
2. WiFi router on shoreside with wireless link to Shoreside and Boat

@whymarrh  pick your choice and I will add hardware as necessary.

Both of these options would need to be configured such that the connection is established upon startup and we are able to SSH into the Pi without any intervention with the boat, simply add power.

whymarrh, 2016-12-16:
No. 1 makes the most sense:

> 1. WiFi router onboard boat with wired link to Pi and wireless link to Shoreside

whymarrh, 2016-12-17:
Is there something in particular we want to track via this issue? I am thinking this can be closed in favour of zero or more issues in [LakeMaps/pi](https://github.com/LakeMaps/pi). That is, if we use the image from there and attach it to a router, we're pretty much good to go.

arandell93, 2016-12-17:
I just want to make sure the process is automatic. That is, we power up and the pi is automatically connected to the network **with a static IP**. Then, from any computer which connects via WiFi we can simply SSH into the pi to do any required control stuff.

If that is easily done with the image you have built, go ahead and close.

whymarrh, 2016-12-17:
I am opposed to using a static IP unless there is a way in the router settings to map a MAC address to a static IP. Having the Pi only attach to the network using a static IP just makes debugging difficult.

arandell93, 2016-12-17:
Noob question: How will we know where to SSH? Do we just poll the subnet until we see the pi and then SSH to there? Or can we SSH some way besides explicitly stating the IP, rather a device name or something and it just finds it?

whymarrh, 2016-12-17:
Both of those work but you can also see a list of devices on the network by going to 192.168.1.1.

arandell93, 2016-12-17:
Good enough. Close 'er up.

closed 2016-12-17 by whymarrh