LakeMaps/boat #15: Wireless Module Message Error Handling

arandell93, 2016-12-14:
During testing of Rev 1.0 it was determined that there exist instances of communication with the wireless module that the boat is unable to handle correctly. This can lead to errors in parsing messages.

See [Wireless Module Wiki](https://github.com/LakeMaps/microcontrollers/wiki/Wireless-Communication-Microcontroller-API) for details on all messages to/from the Wireless Module.

whymarrh, 2017-01-14:
Do you remember any more details about this? What errors occurred and what needs to be fixed?

arandell93, 2017-01-14:
If you recall, I think we had issues with the syncing of the 61 bytes and determining if a message was valid.

Also, I suspect a similar garbling of wireless comms is what occurred to cause the boat to stop working the first time in the pond, but since we didn't log the readout we have no way of knowing for sure.

Essentially, we just want to be able to handle a situation from either end where the message doesn't look right and be able to just move on without crashing by logging an error and just moving to the next message.

whymarrh, 2017-01-14:
Alright, malformed messages are handled appropriately in the new message parser on the boat (#19)

closed 2017-01-14 by whymarrh