LakeMaps/boat #16: Develop Scripts for Shut Down and Restart of Pi via GPIO status

arandell93, 2016-12-14:
In order to provide a method for safetly shutting down or restarting the pi, we need the ability to initiate a safe shutdown from the outside of the boat without needing to be communicating with it from Shoreside.

This is best implemented using buttons attached to the GPIO pins. I will provide hardware required for this.

Proposal:

1. One GPIO pin to initiate a shutdown script
2. One GPIO pin to initiate a restart script

arandell93, 2016-12-16:
See this [link](https://www.element14.com/community/docs/DOC-78055/l/adding-a-shutdown-button-to-the-raspberry-pi-b) from the element14 community.

I am assigning 

- GPIO 23 as the shutdown button
- GPIO 24 as the restart button

arandell93, 2016-12-16:
Note: GPIO pins must be configured with internal pull-up resistances enabled, like so:

`# Use the Broadcom SOC Pin numbers`
`# Setup the Pins with Internal pullups enabled and PINs in reading mode.` 
`GPIO.setmode(GPIO.BCM)` 
`GPIO.setup(23, GPIO.IN, pull_up_down = GPIO.PUD_UP)`
`GPIO.setup(24, GPIO.IN, pull_up_down = GPIO.PUD_UP)`

whymarrh, 2016-12-17:
Can we have some sort of indicator (an LED?) onboard that can let us know the process has begun?

arandell93, 2016-12-17:
Yeah we can, as described [here](https://thepihut.com/blogs/raspberry-pi-tutorials/27968772-turning-on-an-led-with-your-raspberry-pis-gpio-pins).

We will use

- GPIO 4 (pin 7) for this purpose.

I propose a different blinking pattern depending on what script (reset or shutdown) you run. Perhaps solid for shutdown and blink for reset?

whymarrh, 2017-01-17:
> I propose a different blinking pattern depending on what script (reset or shutdown) you run. Perhaps solid for shutdown and blink for reset?

That is, I would set pin 7 high during shutdown and it would turn off when the Pi powers off?

For reset, some odd pattern of highs and lows (e.g. (high + low)\* + high + off)?

arandell93, 2017-01-17:
Yessir!

whymarrh, 2017-01-24:
> I propose a different blinking pattern depending on what script (reset or shutdown) you run. Perhaps solid for shutdown and blink for reset?

Let's not be at that. How about using:

```
Name:   gpio-poweroff
Info:   Drives a GPIO high or low on poweroff (including halt)
Load:   dtoverlay=gpio-poweroff,<param>=<val>
Params: gpiopin                 GPIO for signalling (default 26)

        active_low              Set if the power control device requires a
                                high->low transition to trigger a power-down.
                                Note that this will require the support of a
                                custom dt-blob.bin to prevent a power-down
                                during the boot process, and that a reboot
                                will also cause the pin to go low.
```

from the [`gpio-poweroff` overlay](https://github.com/raspberrypi/linux/blob/rpi-4.4.y/arch/arm/boot/dts/overlays/README#L377-L387). We can drive GPIO 4 low->high when powering off. Thoughts?

arandell93, 2017-01-24:
Won't that mean that the LED will only light up for a brief period of time as the unit is powering down, and will do the same thing regardless of reset or shutdown?

whymarrh, 2017-01-29:
Thinking about it more, I think what I proposed above should be fine for our purposes. GPIO 4 will be high long enough for us to notice it. We can use that pin in addition to the red and green lights onboard the Pi itself (it's not like it's underwater 😏). I think having that information will be more than enough to determine if the Pi is responding to the buttons.

As for knowing when the Pi is back in action, if I were to schedule a light or something to notify us, it'd be scheduled for the exact same time the control software starts up, at which point we may as well use any number of other factors to know whether it's working. If we really want some indication that the Pi is up and running, we can open up a new issue to add something to the control software proper.

arandell93, 2017-01-29:
> We can use that pin in addition to the red and green lights onboard the Pi itself

Absolutely fine for now but when we change enclosures there are no guarantees that it will be transparent. The LED is to be mounted in the enclosure lid to give us that indication at that point.

> we may as well use any number of other factors

Such as? Also we can divert this to a new issue if you would like, but we'll push it to the next release post 1.2 bug fixes. Cool?

whymarrh, 2017-01-29:
One possible example would be the the boat sending its position to the shore—no positions, no boat.

I admit it's not foolproof—if we want a "hardware" indication, we can have the control software do something (I'd be open to whatever, even if it's another LED) though I think that merits its own issue.

arandell93, 2017-01-29:
Moving indicator LED discussion to #24

closed 2017-01-29 by arandell93