LakeMaps/boat #17: Investigate Potential Sources of Latency

arandell93, 2016-12-14:
More testing is required to determine the sources of latency within the controller input - motor output chain.

It would be beneficial to quantify the latency as best as possible, then vary individual parts of the control system to determine the affect they have on overall system latency.

Brainstorm possible sources here and we can determine the most likely sources in order to address those first.

arandell93, 2016-12-16:
I will investigate and quantify the time elapsed between sending a speed value to the Propulsion Module. (Ref [microcontrollers/issues/6](https://github.com/LakeMaps/microcontrollers/issues/6))

whymarrh, 2017-01-28:
I'm closing this seeing how LakeMaps/microcontrollers#6 and LakeMaps/microcontrollers#7 are closed.

closed 2017-01-28 by whymarrh