LakeMaps/boat #18: Implement GPS Parser

arandell93, 2016-12-21:
For Rev 1.2 we will be adding GPS and implementing its functionality over time. We will likely be using the [Adafruit Ultimate GPS Breakout](https://www.adafruit.com/product/746) or similar. Adafruit provides a page with a [tutorial](https://learn.adafruit.com/adafruit-ultimate-gps-on-the-raspberry-pi/introduction) for using it with the Pi if you need something to get started.

The boat requires a library to parse the [NMEA 0183](https://en.wikipedia.org/wiki/NMEA_0183) sentences being received from the GPS module. Page 18 of the [GPS datasheet](https://cdn-shop.adafruit.com/datasheets/GlobalTop-FGPMMOPA6H-Datasheet-V0A.pdf) lists the sentences that the GPS will send. It is up to us to parse the information and use it as we see fit.

Additionally, we can configure the modules to provide faster update rates, select data, etc. Information about configuring the module can be found [here](https://cdn-shop.adafruit.com/datasheets/PMTK_A11.pdf).

arandell93, 2016-12-21:
We are only required to interface to four of the GPS pins (Vin, GND, TX, RX). [The others are optional](https://learn.adafruit.com/adafruit-ultimate-gps/pinouts).
[image]

GPS will be powered directly from the Pi and will use the following pins:

- Pin 4 - +5 V to GPS
- Pin 6 - GND to GPS
- Pin 8 - TX to GPS
- Pin 10 - RX from GPS

[image]

whymarrh, 2017-01-10:
refs #5?

arandell93, 2017-01-12:
> refs #5?

Similar, but that issue is obsolete. This issue is more appropriate since both the hardware and the interface has changed.

arandell93, 2017-01-25:
As discussed yesterday, we toyed with the idea of using the [Sparkfun GPS Shield](https://learn.sparkfun.com/tutorials/gps-shield-hookup-guide) (discontinued and info not available on Sparkfun's website) that I already have as a temporary solution for Rev 1.2. However, this would require acquiring an additional Arduino Uno (~$30) on the boat which we do not have.

In light of this, the decision has been made to go ahead and purchase the [Adafruit GPS Breakout](https://www.adafruit.com/product/746) ($40 USD). Currently, these are out of stock on Adafruit's website so I have come up with a solution that will allow us to implement and test a GPS module but will not allow us to run the boat with it.

For the time being we can use the Sparkfun board noted above with an Arduino as a 'development platform'. It will behave exactly the same as the Adafruit module so we can confirm the functionality of the software interface on both the boat (this issue) and shoreside (LakeMaps/hangashore#20).

For the boat, the requirements of the GPS software for Rev 1.2 are as follows:

- Parse NMEA 0183 sentences** defined in the [GPS User Manual](https://cdn-shop.adafruit.com/datasheets/GlobalTop-FGPMMOPA6H-Datasheet-V0A.pdf) (page 18) and store in a database
  - [GGA](http://www.gpsinformation.org/dale/nmea.htm#GGA)
  - [GSA](http://www.gpsinformation.org/dale/nmea.htm#GSA)
  - [GSV](http://www.gpsinformation.org/dale/nmea.htm#GSV)
  - [RMC](http://www.gpsinformation.org/dale/nmea.htm#RMC)
  - [VTG](http://www.gpsinformation.org/dale/nmea.htm#VTG)
- Determine whether the data is a valid position or not
- Extract data from the NMEA sentences and package as a payload to be sent over wireless during next interaction
  - Latitude (decimal degrees, convert as required)
  - Longitude (decimal degrees, convert as required)
  - Speed (converted to meters per second)
  - Heading (in integer degrees)
  - [HDOP](https://en.wikipedia.org/wiki/Dilution_of_precision_(navigation))

** Note that while there are 5 types of sentences that the GPS will be sending, the information that will ultimately be used by the system at the current time is contained in just two, GGA and VTG. For Rev 1.2 parsing just these two sentences while ignoring the rest would be acceptable as long as it can easily be expanded once we being using the GPS for autonomous navigation.

arandell93, 2017-02-02:
GPS modules are back in stock. Will order two of them today, one for boat and one for shoreside.

arandell93, 2017-02-02:
Damn, they're out of stock again in less than 15 hours.

arandell93, 2017-02-03:
While we are waiting for the GPS modules to get back in stock I did manage to get the EM406 Sheild/Arduino Uno working as a pass-through NMEA sentence device ([4083a79](https://github.com/LakeMaps/microcontrollers/commit/4083a79456e4ff48ab97b5400081fd0f013b890c)). 

Tested with [SASPlanet](https://www.openhub.net/p/sasplanet) and confirmed it is working:
[image]

whymarrh, 2017-02-14:
Any update on when this hardware will be ready?

arandell93, 2017-02-14:
Ready to test with now.

whymarrh, 2017-02-16:
Alrighty this is what I'll work on next. A few questions regarding the requirements you listed above:

> For the boat, the requirements of the GPS software for Rev 1.2 are as follows:
>
> - Parse NMEA 0183 sentences** defined in the [GPS User [...]

Specifically the determination whether the data is a valid position or not, how does one do that?

arandell93, 2017-02-16:
If you look at the wiki page for [Dilution of Precision (Navigation) -> Meaning of DOP Values](https://en.wikipedia.org/wiki/Dilution_of_precision_(navigation)#Meaning_of_DOP_Values) it gives a table which can be used as a rough guideline when making decisions based on DOP. It's a mess of statistical analysis from what I can tell, but the GPS handles the hard bits for us so we just need to know how to interpret and act based on what the GPS tells us.

HDOP is defined as:

> A measure of the geometric quality of a GPS satellite configuration in the sky. HDOP is a factor in determining the relative accuracy of a horizontal position. The smaller the DOP number, the better the geometry.

Specifically, I think for our purposes our control system will need to wait (i.e. no motor outputs) for the GPS to 'warm up'; to connect to enough satellites to provide an HDOP value _lower_ than _x_. This will mean that we can expect our +/- range (expressed in meters) to be (theoretically) less than _y_ meters. We should still store all GPS data sets (perhaps storing the raw sentences?) but only use sets where the HDOP (or last received HDOP, as HDOP is only sent every few messages) meets our criteria as control data. 

I think it is best for us to be able to change the HDOP required to complete a survey via the GUI so we can do a bit of trial and error to determine what is a typical 'good' number. Let's use an HDOP of 5 as a baseline.

I should also note that before the GPS acquires enough satellites (3) to get a fix, the NMEA sentences will still be sent, but they will not contain data. You would see several commas in a row indicating skipped fields similar to this (modified from the GGA example [here](http://www.gpsinformation.org/dale/nmea.htm#GGA)):

`$GPGGA,123519,,,,,0,01,,,,,,,*47`

The commas are always used to separate data fields so you should be able to skip right along through the message using them, ignoring fields which are missing. The [GGA](http://www.gpsinformation.org/dale/nmea.htm#GGA) message specifically has its own indicator of 'fix quality' which can also be checked to determine validity. [GSA](http://www.gpsinformation.org/dale/nmea.htm#GSA) uses a '3D Fix' field for a similar purpose.

whymarrh, 2017-02-19:
I think the checksums in the following examples from the [GPS User Manual](https://cdn-shop.adafruit.com/datasheets/GlobalTop-FGPMMOPA6H-Datasheet-V0A.pdf) are incorrect:

- `$GPGGA,064951.000,2307.1256,N,12016.4438,E,1,8,0.95,39.9,M,17.8,M,,*65` (p. 19)
- `$GPVTG,165.48,T,,M,0.03,N,0.06,K,A*37` (p. 23)

I think they should be `0x63` and `0x36`, respectively:

- `$GPGGA,064951.000,2307.1256,N,12016.4438,E,1,8,0.95,39.9,M,17.8,M,,*63`
- `$GPVTG,165.48,T,,M,0.03,N,0.06,K,A*36`

If you want, this is how I was calculating them with Python:

```
>>> checksum = lambda s: hex(reduce(lambda acc, c: acc ^ c, map(ord, list(s)), 0))
>>> checksum("GPGGA,064951.000,2307.1256,N,12016.4438,E,1,8,0.95,39.9,M,17.8,M,,")
'0x63'
>>> checksum("GPVTG,165.48,T,,M,0.03,N,0.06,K,A")
'0x36'
```

arandell93, 2017-02-22:
$GPRMC,000850.074,V,,,,,,,150209,,,N*4C
                                       
$GPGGA,000851.074,,,,,0,00,,,M,0.0,M,,0000*59
                                 
$GPGSA,A,1,,,,,,,,,,,,,,,*1E
                                                  
$GPGSV,1,1,00*79

whymarrh, 2017-02-22:
(@arandell93 I've updated the formatting of your comment.)

whymarrh, 2017-02-25:
On determining if a given position is valid or not, from @arandell93's comment above:

> Specifically, I think for our purposes our control system will need to wait (i.e. no motor outputs) for the GPS to 'warm up'; to connect to enough satellites to provide an HDOP value *lower* than [...]

I think this bit is orthogonal to the GPS implementation and is better brought up when we go about navigating by GPS. As such, this will not be included in #33.

arandell93, 2017-02-25:
10-4, makes good sense.

closed 2017-02-25 by whymarrh

arandell93, 2017-03-04:
FYI, the [receiver](https://cdn-shop.adafruit.com/datasheets/PMTK_A11.pdf) on the [Adafruit Board](https://www.adafruit.com/products/746) has settings that can be modified to set things like update rate of GPS (1, 5, 10 Hz) ([page 8](https://cdn-shop.adafruit.com/datasheets/PMTK_A11.pdf)) and the baud rate of the serial line (up to 115200) ([page 9](https://cdn-shop.adafruit.com/datasheets/PMTK_A11.pdf)). 

Factory default settings can be found [on page 16](https://cdn-shop.adafruit.com/datasheets/GlobalTop-FGPMMOPA6H-Datasheet-V0A.pdf).

We will need to make these modifications at some point (increase speed to max) in order to optimize the speed of the control system. However, whenever power is removed from the GPS it loses its settings and returns to factory default. So this is something we will have to configure upon startup. From the perspective of the module, this is a firmware re-flash and will reboot the module each time.

PR #33 (2017-02-25): Implement GPS parser
Closes #18

This PR implements a rudimentary GPS parser (limited to GGA, GSA, GSV, RMC, and VTG sentences). The parser here is loosely based off the parser for microcontroller messages but this application of the parser is a bit lax about its parsing. For example, the parser does not care that messages are separated by `0x0D0A`, it will simply skip characters between the end of one message and the start of the next message `0x24`. This parser does also not handle messages that have special encapsulation in them, seeing as none of the sentences needed for #18 had them (that [... cut]