LakeMaps/boat #28: Action on Receipt of Corrupt Messages (Wireless)

arandell93, 2017-02-05:
Refs #27 

When we determine that a message is corrupt, we should ideally do something like the following:

- Ignore the message and do not reply (as #27 implements)
- Log the corrupted message as it was received with a timestamp
- Increment a corrupt message counter
- If there have been _x_ corrupt messages since the last good message (or perhaps if the ratio of good/corrupt messages over the last _y_ messages is below _z_), attempt to send back a warning across the link in reply to the corrupt message

Thoughts?

whymarrh, 2017-02-05:
> - Log the corrupted message as it was received with a timestamp
> - Increment a corrupt message counter

That seems redundant to at first. We can count the number of log messages to derive the 2nd.

> If there have been x corrupt messages since the last good message (or perhaps if the ratio of good/corrupt messages over the last y messages is below z), attempt to send back a warning across the link in reply to the corrupt message.

Oh, that's why we want the 2nd piece of information handy? I don't think this is that simple: what if it's the wireless microcontroller that is sending corrupted data back?

whymarrh, 2017-02-05:
Also, I'm not sure if I'm reading your last bullet correct there, but #27 isn't only for messages coming in across the wireless link but for the microcontroller communication itself (sure, of which messages coming in across the link are a subset). #27 skips corrupted messages from all microcontrollers.

arandell93, 2017-02-05:
> what if it's the wireless microcontroller that is sending corrupted data back?

Valid point... I didn't think of that.

>  #27 skips corrupted messages from all microcontrollers.

I did initially understand your commit as all microcontrollers, but when I created this issue I was only thinking about the link. Now that you remind me though, it's equally valid for all microcontrollers and whatever we choose to do here should be implemented as such.

Further, I think the same logic should be applied on shoreside, but in that case you only have the wireless link.

arandell93, 2017-02-05:
> what if it's the wireless microcontroller that is sending corrupted data back?

Further to this point, we may be able to use the 'shitty' arduino serial comms connection reset to our advantage here. Perhaps if we get this issue the first thing we try is disconnecting the port and reconnecting?

whymarrh, 2017-02-06:
Thinking about it a bit, I'm not sure what resetting the connection would solve here. That idea, while clever, seems to fall into the category of "wishful thinking." This would have the possibility of resulting in an endless cycle of resetting the connection and continuing to get corrupt messages, no? If we go the route of resetting only once "just to try it," that seems to me like extra state management for no tangible benefit.

whymarrh, 2017-02-06:
Minimally, I think implementing your 2nd bullet (logging the corrupted message as it was received with a timestamp) is a good starting point. (Building on #14.)

arandell93, 2017-02-06:
>  If we go the route of resetting only once "just to try it," that seems to me like extra state management for no tangible benefit.

It is conceivable that the microcontroller could formulate garbage messages that can be fixed with a reset, however I do agree it is a low-probability event. It was just the first (and easiest) thing I could think of to attempt to resolve a high message error rate.

> Minimally, I think implementing your 2nd bullet (logging the corrupted message as it was received with a timestamp) is a good starting point

Agreed, that is definitely the first logical thing to implement. Hopefully if we do see errors, having that log will give us a clue as to where we should look to resolve them rather than attempting to brainstorm and safe-guard against things which may or may not be real issues.

whymarrh, 2017-02-06:
Alright, I should have PR with logging up sometime soon

whymarrh, 2017-02-23:
Closed in favour of #34.

closed 2017-02-23 by whymarrh