LakeMaps/boat #32: Error constructing dynamic protobuf messages

whymarrh, 2017-02-18:
When constructing dynamic protobuf message schemas, the byte array passed to [`FileDescriptorSet#parseFrom(byte[])`](https://developers.google.com/protocol-buffers/docs/reference/java/com/google/protobuf/DescriptorProtos.FileDescriptorSet.html#parseFrom-byte:A-) causes an [`InvalidProtocolBufferException`](https://developers.google.com/protocol-buffers/docs/reference/java/com/google/protobuf/InvalidProtocolBufferException) to be thrown.

whymarrh, 2017-02-18:
I'll try to take a look at this tomorrow—it will be a blocker for our next bench test.

whymarrh, 2017-02-19:
I will hopefully take a look at his before Tuesday.

arandell93, 2017-02-19:
When you said a blocker for our next bench test, did you mean to finish troubleshooting 1.0.1 or did you mean to move past that?

whymarrh, 2017-02-19:
Rev 1.0.1, yeah.

whymarrh, 2017-02-26:
🎉 🎉 🎉 🎉 Woo! Only 6 days off on my time estimate! 🎉 🎉 🎉 🎉

PR #37 (2017-02-26): Fix error from constructing dynamic protobuf messages
Closes #32

This PR fixes the  [`InvalidProtocolBufferException`][1] that was thrown when trying to parse a `FileDescriptorSet` (using [`FileDescriptorSet#parseFrom(byte[])`][2]). I've gone ahead an replaced all of that with the "more proper" use of protobufs—`protoc` is now used to compile [the schemas](https://github.com/LakeMaps/schemas) and the generated classes are used for serialization.

  [1]:https://developers.google.com/protocol-buffers/docs/reference/java/com/google/protobuf/InvalidProtocolBufferException
  [2]:https://developers [... cut]
closed 2017-02-26 by whymarrh