LakeMaps/boat #40: Method for Updating GPS Module Settings

arandell93, 2017-03-04:
Refs #33 and #18 

FYI, the [receiver](https://cdn-shop.adafruit.com/datasheets/PMTK_A11.pdf) on the [Adafruit Board](https://www.adafruit.com/products/746) has settings that can be modified to set things like update rate of GPS (1, 5, 10 Hz) ([page 8](https://cdn-shop.adafruit.com/datasheets/PMTK_A11.pdf)) and the baud rate of the serial line (up to 115200) ([page 9](https://cdn-shop.adafruit.com/datasheets/PMTK_A11.pdf)). 

Factory default settings can be found [on page 16](https://cdn-shop.adafruit.com/datasheets/GlobalTop-FGPMMOPA6H-Datasheet-V0A.pdf).

We will need to make these modifications at some point (increase speed to max) in order to optimize the speed of the control system. However, whenever power is removed from the GPS it loses its settings and returns to factory default. So this is something we will have to configure upon startup. From the perspective of the module, this is a firmware re-flash and will reboot the module each time.

whymarrh, 2017-03-04:
> So this is something we will have to configure upon startup. From the perspective of the module, this is a firmware re-flash and will reboot the module each time.

Control software startup? Which module?

whymarrh, 2017-03-04:
Also, could I bother you to ELI5 the difference between the [PMTK_A11](https://cdn-shop.adafruit.com/datasheets/PMTK_A11.pdf) and the [FGPMMOPA6H](https://cdn-shop.adafruit.com/datasheets/GlobalTop-FGPMMOPA6H-Datasheet-V0A.pdf)

arandell93, 2017-03-04:
> Control software startup?

Correct. When the Pi starts up, before it is 'ready' to do boat things it needs to first configure the GPS.

> Which module?

GPS Module will reboot upon a configuration change. I assume this will happen each time, so if we set 3 settings it will reboot 3 times [citation needed].

> [C]ould I bother you to ELI5 the difference between the PMTK_A11 and the FGPMMOPA6H

From [FGPMMOPA6H](https://cdn-shop.adafruit.com/datasheets/GlobalTop-FGPMMOPA6H-Datasheet-V0A.pdf) page 4
> The FGPMMOPA6H utilizes the MediaTek new generation GPS Chipset MT3339 

From [PMTK_A11](https://cdn-shop.adafruit.com/datasheets/PMTK_A11.pdf) page 6

> Packet Type: 001 PMTK_ACK
Support Chip Type:
MT3318、MT3329、MT3339

TL;DR: PMTK is a set of NMEA command messages that is supported by MT3339 chipset GPS modules. The FGPMMOPA6H is the name of the GPS hardware which is using said chipset.

arandell93, 2017-04-30:
Info:
https://cdn-shop.adafruit.com/datasheets/PMTK_A08.pdf

Set the GPS Update Rate to 10 Hz
(Tested and works but must be at a sufficient NMEA baud such as 57600)
$PMTK220,100*2F

Set NMEA Baud Rate to 57600 bit/s
(Tested and works)
$PMTK251,57600*2C

Set rate of position fixing activity to 300 ms
(Not working?)
$PMTK300,300,0,0,0,0*0A

whymarrh, 2017-05-04:
Did we confirm that the GPS does/doesn't keep its settings across serial (re)connections?

whymarrh, 2017-05-04:
Am I correct in that we need to set the baud rate to 57600 bps *before* setting the update rate to 10Hz?

arandell93, 2017-05-10:
> Did we confirm that the GPS does/doesn't keep its settings across serial (re)connections?

It only resets upon power cycling. Serial reconnects are fine.

> Am I correct in that we need to set the baud rate to 57600 bps before setting the update rate to 10Hz?

You are correct.

PR #107 (2017-06-23): Implement hardware serial GPS
Closes #40
Closes #90

This PR updates the control software for the new GPS receiver. All the sentence parsing is the same.

At start-up we now:

1. Connect to the receiver using 9600 baud
1. Set the NMEA baud rate to 57600 (`PMTK_SET_NMEA_BAUDRATE`)
1. Disconnect and reconnect using 57600 baud
1. Set the NMEA update rate to 10 Hz (`PMTK_SET_NMEA_UPDATERATE`)
closed 2017-06-23 by whymarrh