LakeMaps/boat #47: Control latency diagnosis

whymarrh, 2017-03-10:
Alrighty, after a bit of time to think about it, let's try and tackle this. I've created a project for [Revision 1.0.1 Control Latency Diagnosis](https://github.com/orgs/LakeMaps/projects/6) which we can use to track the issues across repositories and we can use this issue both as a tracking issue and a place for **on-topic** discussion.

We can create issues as needed in the respective repositories (or not, if the task is small enough) once we agree on the specifics. As of right now, to be able to quantify/estimate the complete control latency, the following items need to be quantified:

- Hangashore
    - [ ] Time to poll the gamepad
    - [ ] Time to process a motion
    - [ ] Time to send a value to a microcontroller
- Boat
    - [ ] Time to process a motion
    - [ ] Time to send a value to a microcontroller
- Microcontrollers
    - [ ] Time to adjust propulsion (to set speed)
    - [ ] Time to send a value across the wireless link
        - We will need to assume that the time to send a value across the wireless link minus the theoretical wireless link transfer time is the processing overhead (only some of which is under our control)

I'll update the list with issue numbers as they're created.

whymarrh, 2017-03-10:
Additionally we need to manage expectations. IIRC, we talked about our ideal control latency being 100ms and...yeah that's smaller than the theoretical wireless link latency.<sup>\[1\]</sup>

1\. Which, according to LakeMaps/microcontrollers#7, is 101.7 ms

arandell93, 2017-03-11:
Yeah I recognize 100 ms is not realistic.  I standby my statement of 0.5 seconds being manageable, with less preferred. Recall that we had in the realm of 150-200 ms latency on the ROV and it was unnoticeable under almost all circumstances. Vehicle control latency is important to distinguish form something like visual latency (such as when interacting with a computer and seeing letters appear on a screen as you type, or movements in a video game as you play). 

The crispness of the control and system will affect the human's expected latency. By crispness I am referring to the expectation of an immediate action to occur upon input. For example, when you press the space bar, you know that there should be a whitespace character appear on your screen immediately. However, if you consider the act of driving a car and pressing the gas pedal, the driver doesn't expect the speed to instantly change due to the time it takes for a car's engine to rev up, the transmission to change gears, and other vehicle dynamics. 

If our case, one does not expect a boat to be able to change direction on a dime as it has momentum moving through a viscous medium, has propellor slip, etc. This is similar to the ROV, but I would argue the boat is intuitively more 'sluggish' to respond.

arandell93, 2017-03-14:
Just thought of this:

Refs LakeMaps/microcontrollers/#20 and LakeMaps/microcontrollers/#7 

Since the 61 byte packet uses addressing and encryption, who is to say that what is actually being transferred over the link (at 4.8 kbit/s) is not the full 255 bytes representing the address and the encrypted version of the 61 byte packet?

More investigation needed to determine whether or not this is the case. If so, we may have a lock on what could be the majority of our inherent latency. (255 bytes at 4800 bit/s is 0.425 seconds).

FifoIronton, 2017-03-14:
I tried looking into this a little bit yesterday... 

I _don't think_ that AES encryption adds any extra bits to the encrypted data, instead just doing some weird math to substitute one byte for another byte. In AES, blocks of 16 bytes of plaintext are turned into 16 bytes of cyphertext. A cool looking interactive demonstration of it is found [here](http://www.formaestudio.com/rijndaelinspector/archivos/Rijndael_Animation_v4_eng.swf).

According to the datasheet, the address should only be one byte long, so that also shouldn't add significant time to the wireless transfer.

This being said, it would be interesting to see if a latency change is seen without the encryption

whymarrh, 2017-03-14:
> This being said, it would be interesting to see if a latency change is seen without the encryption

I agree. Do we know if the wireless link has hardware dedicated to encryption?

whymarrh, 2017-03-14:
I have started to add benchmarks using [JMH](http://openjdk.java.net/projects/code-tools/jmh/) in #50 to attach numbers to some of the items in the OP.

**Disclaimer**: I am not 100% confident that these are _the most correct_ numbers. [Benchmarking is hard](https://www.ibm.com/developerworks/java/library/j-jtp02225/index.html) and I have tried my best to ensure that these numbers are not _incorrect_. I am confident that these numbers are NOT five (5) orders of magnitude off though.

With that said, this is what I have so far:

```
# JMH 1.18 (released 1 days ago)
# VM version: JDK 1.8.0_121, VM 25.121-b15
# VM invoker: /opt/zulu8.20.0.5-jdk8.0.121/jre/bin/java
# VM options: <none>
# Warmup: <none>
# Measurement: 20 iterations, 1 s each
# Timeout: 10 min per iteration
# Threads: 1 thread, will synchronize iterations
# Benchmark mode: Average time, time/op
# Benchmark: jmh.BoatBenchmark.tick

# Run progress: 0.00% complete, ETA 00:02:00
# Fork: 1 of 1
Iteration   1: 0.018 ms/op
Iteration   2: 0.005 ms/op
Iteration   3: 0.004 ms/op
Iteration   4: 0.003 ms/op
Iteration   5: 0.003 ms/op
Iteration   6: 0.003 ms/op
Iteration   7: 0.003 ms/op
Iteration   8: 0.003 ms/op
Iteration   9: 0.004 ms/op
Iteration  10: 0.004 ms/op
Iteration  11: 0.004 ms/op
Iteration  12: 0.003 ms/op
Iteration  13: 0.003 ms/op
Iteration  14: 0.003 ms/op
Iteration  15: 0.003 ms/op
Iteration  16: 0.003 ms/op
Iteration  17: 0.003 ms/op
Iteration  18: 0.003 ms/op
Iteration  19: 0.003 ms/op
Iteration  20: 0.003 ms/op

Result "jmh.BoatBenchmark.tick":
  0.004 ±(99.9%) 0.003 ms/op [Average]
  (min, avg, max) = (0.003, 0.004, 0.018), stdev = 0.003
  CI (99.9%): [0.001, 0.007] (assumes normal distribution)

# JMH 1.18 (released 1 days ago)
# VM version: JDK 1.8.0_121, VM 25.121-b15
# VM invoker: /opt/zulu8.20.0.5-jdk8.0.121/jre/bin/java
# VM options: <none>
# Warmup: <none>
# Measurement: 20 iterations, 1 s each
# Timeout: 10 min per iteration
# Threads: 1 thread, will synchronize iterations
# Benchmark mode: Average time, time/op
# Benchmark: jmh.MicrocontrollerBenchmark.receiveWirelessLinkMicrocontroller

# Run progress: 16.67% complete, ETA 00:01:44
# Fork: 1 of 1
Iteration   1: 0.009 ms/op
Iteration   2: 0.004 ms/op
Iteration   3: 0.004 ms/op
Iteration   4: 0.004 ms/op
Iteration   5: 0.004 ms/op
Iteration   6: 0.004 ms/op
Iteration   7: 0.004 ms/op
Iteration   8: 0.004 ms/op
Iteration   9: 0.004 ms/op
Iteration  10: 0.004 ms/op
Iteration  11: 0.004 ms/op
[... 169 lines cut ...]
  CI (99.9%): [0.001, 0.002] (assumes normal distribution)

# Run complete. Total time: 00:02:04

Benchmark                                                    Mode  Cnt  Score    Error  Units
BoatBenchmark.tick                                           avgt   20  0.004 ±  0.003  ms/op
MicrocontrollerBenchmark.receiveWirelessLinkMicrocontroller  avgt   20  0.004 ±  0.001  ms/op
MicrocontrollerBenchmark.resetPropulsionMicrocontroller      avgt   20  0.001 ±  0.001  ms/op
MicrocontrollerBenchmark.resetWirelessLinkMicrocontroller    avgt   20  0.001 ±  0.001  ms/op
MicrocontrollerBenchmark.sendWirelessLinkMicrocontroller     avgt   20  0.004 ±  0.001  ms/op
MicrocontrollerBenchmark.setSpeedPropulsionMicrocontroller   avgt   20  0.001 ±  0.001  ms/op
```

([I've posted a gist with a run with microsecond accuracy][1] and [the tables for both millisecond and microsecond output][2].)

To explain the benchmark names:

- `BoatBenchmark.tick` is the time it takes to process one motion value once it is in the system. This includes calculating the prop speeds from the motion value.
- `MicrocontrollerBenchmark.receiveWirelessLinkMicrocontroller` is the time it takes to receive a 64-byte message from the wireless link **sans-I/O**. This does not include the time it takes to read from a real serial device. This does include parsing and constructing messages.
- `MicrocontrollerBenchmark.setSpeedPropulsionMicrocontroller` is the time it takes to send the set speed command to a hypothetical microcontroller. Again, this is **sans-I/O** and does not include the time it takes to write to a real serial device. This does include constructing the serial buffer.

All of that says what? In my opinion, this shows me that these particular sections of the overall system are negligible contributors to overall system latency. The combined latency of a wireless recv, boat tick, and prop send is at most 1 ms.

  [1]:https://gist.github.com/whymarrh/068eda2010eafc5ea821e519a625165e
  [2]:https://gist.github.com/whymarrh/8b3ad315a57d94570deadbde5dd29d9e

whymarrh, 2017-03-14:
Between a wireless recv and a boat tick is deserializing a motion value, which still needs a number.

arandell93, 2017-03-15:
So far in determining the time to send a wireless message across the link :

- Execution of Transmit command (between send and receive response from microcontroller). Average of 11 tests: 83.4 ms*

- Execution of Receive command (between send and receive response from microcontroller). Average of 11 tests: 71.1 ms

- Time to send across the wireless link: Not determined.

*Note that this is not necessarily the time incurred by the system, it is simply the time it takes for the shoreside to receive a response confirming transmission after sending a payload to be transmitted. By comparison, sending a null message (which returns an error message from the microcontroller) takes (average of 11 tests) 48.8 ms. The difference (83.4 - 48.8 = 34.6) can be accounted for as the extra time to send a longer message over serial and the difference in time taken to step through the 'transmit command' loop vs. the 'error' loop. The transmit command loop includes a for loop which pushes the data into the array for transmission and calling `radio.sendWithRetry` from the [LowPowerLab RFM69 library](https://github.com/LowPowerLab/RFM69). However, I have yet to determine whether or not the `radio.sendWithRetry` function finishes sending data over the wireless message before returning with the acknowledged flag, so there could be added time there that is not accounted for.

FifoIronton, 2017-03-15:
> Do we know if the wireless link has hardware dedicated to encryption?

I poked through the datasheet a little, and found this interesting section on pages 56 and 57:

> 5.5.5.1. Tx Processing 
[...]
On Tx command the Packet handler state machine takes over the control and If encryption is enabled then the message inside the FIFO is read in blocks of 16 bytes (padded with 0s if needed), encrypted and stored back to FIFO.
All this processing is done in Tx mode before enabling the packet handling state machine. Only the Message part of the packet is encrypted and preamble, sync word, length byte, address byte and CRC are not encrypted. 

> 5.5.5.2. Rx Processing
[...]
Once the complete packet has been received. The data is read from the FIFO, decrypted and written back to FIFO.
The PayloadReady interrupt is issued once the decrypted data is ready in the FIFO for reading via the SPI interface.

> The AES encryption/decryption cannot be used on the fly i.e. while transmitting and receiving data. Thus when AES encryption/decryption is enabled, the FIFO acts as a simple buffer. This buffer is filled before initiating any transmission.
The data in the buffer is then encrypted before the transmission can begin. On the receive side the decryption is initiated only once the complete packet has been received in the buffer.
The encryption/decryption process takes approximately 7.0 us per 16-byte block. Thus for a maximum of 4 blocks (i.e. 64 bytes) it can take up to 28 us for completing the cryptographic operations.
The receive side sees the AES decryption time as a sequential delay before the PayloadReady interrupt is available.
The Tx side sees the AES encryption time as a sequential delay in the startup of the Tx chain, thus the startup time of the Tx will increase according to the length of data. 

TLDR: There is some hardware that encrypts 16 byte blocks in the buffer and then sticks it back in the buffer before sending/officially receiving it. It takes a few orders of magnitude less time than our other latency sources

whymarrh, 2017-03-16:
More numbers incoming:

```
# JMH 1.18 (released 2 days ago)
# VM version: JDK 1.8.0_121, VM 25.121-b15
# VM invoker: /opt/zulu8.20.0.5-jdk8.0.121/jre/bin/java
# VM options: <none>
# Warmup: <none>
# Measurement: 20 iterations, 1 s each
# Timeout: 1 min per iteration
# Threads: 1 thread, will synchronize iterations
# Benchmark mode: Average time, time/op
# Benchmark: jmh.BoatBenchmark.core

# Run progress: 0.00% complete, ETA 00:00:20
# Fork: 1 of 1
Iteration   1: 0.396 ms/op
Iteration   2: 0.154 ms/op
Iteration   3: 0.112 ms/op
Iteration   4: 0.064 ms/op
Iteration   5: 0.052 ms/op
Iteration   6: 0.050 ms/op
Iteration   7: 0.049 ms/op
Iteration   8: 0.062 ms/op
Iteration   9: 0.051 ms/op
Iteration  10: 0.050 ms/op
Iteration  11: 0.050 ms/op
Iteration  12: 0.053 ms/op
Iteration  13: 0.051 ms/op
Iteration  14: 0.050 ms/op
Iteration  15: 0.051 ms/op
Iteration  16: 0.053 ms/op
Iteration  17: 0.055 ms/op
Iteration  18: 0.057 ms/op
Iteration  19: 0.058 ms/op
Iteration  20: 0.070 ms/op

Result "jmh.BoatBenchmark.core":
  0.079 ±(99.9%) 0.068 ms/op [Average]
  (min, avg, max) = (0.049, 0.079, 0.396), stdev = 0.079
  CI (99.9%): [0.011, 0.148] (assumes normal distribution)

# Run complete. Total time: 00:00:20

Benchmark           Mode  Cnt  Score   Error  Units
BoatBenchmark.core  avgt   20  0.079 ± 0.068  ms/op
```

This time on *What Do These Numbers Mean* we're again quantifying the amount of time it takes to process a motion value in the core of the boat's control software. Note that this is still sans-I/O as we're not actually connected to a serial device. This here does include deserializing a motion value though.

We're still < 1ms.

whymarrh, 2017-03-21:
Related:

- [Investigate Latency Introduced by Propulsion Module](https://github.com/LakeMaps/microcontrollers/issues/6)
- [Investigate Latency Introduced by Wireless Link](https://github.com/LakeMaps/microcontrollers/issues/7)

whymarrh, 2017-03-25:
With #57 merged, the new control latency will be quantified through #51.

whymarrh, 2017-04-07:
The final\* latency numbers produced from #51:

| Send Timestamp (`"%S.%f"`) | Recv Timestamp (`"%S.%f"`) | Δ*t* (ms) | ½Δ*t* (ms) |
|----------------------------|----------------------------|-----|------|
| 03.800                     | 04.252                     | 452 | 226  |
| 04.263                     | 05.575                     | 312 | 156  |
| 06.473                     | 06.608                     | 135 | 67   |
| 06.925                     | 07.106                     | 181 | 90   |
| 06.982                     | 07.170                     | 188 | 94   |
| 06.859                     | 07.041                     | 182 | 91   |
| 07.045                     | 07.250                     | 205 | 102  |
| 07.194                     | 07.374                     | 180 | 90   |
| 08.282                     | 08.420                     | 138 | 69   |
| 08.717                     | 08.853                     | 136 | 68   |
| 09.826                     | 11.177                     | 351 | 175  |
| 09.948                     | 11.627                     | 679 | 339  |
| 47.891                     | 48.412                     | 521 | 260  |
| 48.609                     | 50.603                     | 994 | 497  |
| 23.664                     | 23.871                     | 207 | 103  |
| 24.311                     | 24.463                     | 152 | 76   |
| 25.048                     | 26.129                     | 81  | 40   |
| 26.070                     | 26.236                     | 166 | 83   |
| 30.507                     | 31.222                     | 715 | 357  |
| 31.161                     | 31.268                     | 107 | 53   |
| 31.714                     | 32.277                     | 563 | 281  |
| 32.216                     | 32.595                     | 379 | 189  |
| 32.601                     | 32.768                     | 167 | 83   |
| 32.659                     | 32.961                     | 302 | 151  |
| 32.969                     | 33.192                     | 223 | 111  |
| 33.195                     | 34.015                     | 820 | 410  |
| 33.958                     | 34.145                     | 187 | 93   |
| 32.969                     | 33.192                     | 223 | 111  |
| 33.195                     | 34.015                     | 820 | 410  |
| 33.958                     | 34.145                     | 187 | 93   |
| 34.208                     | 34.516                     | 308 | 154  |
| 35.422                     | 36.057                     | 635 | 317  |
| 36.289                     | 36.640                     | 351 | 175  |
| 37.229                     | 37.578                     | 349 | 174  |
| 38.536                     | 39.793                     | 257 | 128  |
| 39.736                     | 39.898                     | 162 | 81   |
| 40.892                     | 41.449                     | 557 | 278  |
| 41.462                     | 41.605                     | 143 | 71   |
| 41.544                     | 42.057                     | 513 | 256  |
| 41.997                     | 42.355                     | 358 | 179  |
| 41.462                     | 41.605                     | 143 | 71   |
| 41.544                     | 42.057                     | 513 | 256  |
| 41.997                     | 42.355                     | 358 | 179  |
| 43.320                     | 43.471                     | 151 | 75   |
| 43.475                     | 43.796                     | 321 | 160  |
| 43.740                     | 44.050                     | 310 | 155  |
| 44.090                     | 44.481                     | 391 | 195  |
| 44.492                     | 44.681                     | 189 | 94   |
| 44.627                     | 44.750                     | 123 | 61   |
| 44.694                     | 45.086                     | 392 | 196  |

I think this is a useful/valid representation (I'm not a scientician):

[image]

For those interested (and as a note to myself), the diff of the change that produced the logs:

```diff
@@ -16,8 +16,12 @@ const makeWirelessDriver = (name: string) => {
 
         motion$
             .mergeMap((data) => Motion.schema.encode(data))
-            .exhaustMap((motion: Buffer) => Observable.defer(() => m.send(motion).then(_ => m.recv())))
+            .exhaustMap((motion: Buffer) => {
+                console.log(motion.toString(`hex`));
+                return Observable.defer(() => m.send(motion).then(_ => m.recv()))
+            })
             .subscribe((recv) => {
+                console.log(recv.body().toString(`hex`));
                 rssi$.next(recv.rssi());
             });
         return {
```

\* Despite the entire system control latency not being quantified, @arandell93 has proposed putting aside the latency discussion for now and moving onto Rev 1.2.

closed 2017-04-08 by whymarrh

whymarrh, 2017-04-11:
Turns out™ the default bitrate provided by the RFM69 library is 55,555 bps:<sup>[\[1\]][1]</sup>

```cpp
/* 0x03 */ { REG_BITRATEMSB, RF_BITRATEMSB_55555}, // default: 4.8 KBPS
/* 0x04 */ { REG_BITRATELSB, RF_BITRATELSB_55555},
```

Where `0x03` and `0x04` are the registers for the bitrate MSB and LSB, respectively:<sup>[\[2\]][3]</sup>

```cpp
#define REG_BITRATEMSB    0x03
#define REG_BITRATELSB    0x04
```

And where `RF_BITRATEMSB_55555` and `RF_BITRATELSB_55555` are:<sup>[\[3\]][4]</sup>

```cpp
// custom bit rates
#define RF_BITRATEMSB_55555           0x02
#define RF_BITRATELSB_55555           0x40
```

Yes, a bitrate of 55,555 bps isn't listed in [the table on p. 22 of the datasheet](https://cdn-shop.adafruit.com/product-files/3076/sx1231.pdf), but note the text of §3.4.2:

> **3.4.2. Bit Rate Setting**
> When using the SX1231 in Continuous mode, the data stream to be transmitted can be input directly to the modulator via pin 9 (DIO2/DATA) in an asynchronous manner, unless Gaussian filtering is used, in which case the DCLK signal on pin 10 (DIO1/DCLK) is used to synchronize the data stream. See section 3.4.5 for details on the Gaussian filter.
> 
> In Packet mode or in Continuous mode with Gaussian filtering enabled (refer to section 5.5 for details), the Bit Rate (BR) is controlled by bits *BitRate* in *RegBitrate*:
>
> [image]
>
>
> Amongst others, the following Bit Rates are accessible:

And the Chip Specification from §2.4:

> **2.4. Chip Specification**
> The tables below give the electrical specifications of the transceiver under the following conditions: Supply voltage VBAT1= VBAT2=VDD=3.3 V, temperature = 25 °C, *FXOSC* = 32 MHz, F<sub>RF</sub> = 915 MHz, Pout = +13dBm, 2-level FSK modulation without pre-filtering, FDA = 5 kHz, Bit Rate = 4.8 kb/s and terminated in a matched 50 Ohm impedance, unless otherwise specified.

With those two pieces of information, I can confirm that 3.2 × 10^7 / `0x0240` = 55,555,<sup>[\[4\]][5]</sup> which is consistent with the rest of the table on p. 22.

At a bitrate of 55,555 bps transferring 61 bytes would take 8.784 ms.<sup>[\[5\]][6]</sup>

  [1]:https://github.com/LowPowerLab/RFM69/blob/f8192e1fd3c2effdaba97e57522b408336a144c5/RFM69.cpp#L48-L49
  [2]:https://github.com/LowPowerLab/RFM69/blame/f8192e1fd3c2effdaba97e57522b408336a144c5/RFM69.cpp#L48-L49
  [3]:https://github.com/LowPowerLab/RFM69/blob/f8192e1fd3c2effdaba97e57522b408336a144c5/RFM69registers.h#L37-L38
  [4]:https://github.com/LowPowerLab/RFM69/blob/f8192e1fd3c2effdaba97e57522b408336a144c5/RFM69registers.h#L199-L201
  [5]:https://www.wolframalpha.com/input/?i=3.2+×+10%5E7+%2F+0x0240
  [6]:https://www.wolframalpha.com/input/?i=61+bytes+at+55555+bps