LakeMaps/boat #53: Convert position information from GPS into DMS

whymarrh, 2017-03-16:
`GGA` and `RMC` offer latitude and longitude in `ddmm.mmmm`. These should be converted into [DMS](https://en.wikipedia.org/wiki/Decimal_degrees).

From pp. 19 and 22 from the [FGPMMOPA6H datasheet](https://cdn-shop.adafruit.com/datasheets/GlobalTop-FGPMMOPA6H-Datasheet-V0A.pdf):

| Name | Example | Description |
|:---|:---|:---|
| Latitude | 2307.1256 | ddmm.mmmm |
| Longitude | 12016.4438 | dddmm.mmmm |

arandell93, 2017-03-16:
> These should be converted into DMS.

You state Degrees, Minutes, Seconds but link to decimal degrees. I assume you had intended to use decimal degrees? (As it is the superior unit)

whymarrh, 2017-03-16:
¯\\\_(ツ)\_/¯

arandell93, 2017-03-16:
Use decimal degrees.

whymarrh, 2017-04-23:
This was added in 604710d725bc1275defb117895121dfba989f0d9 but will certainly need to "cleaned up" at some point

PR #80 (2017-05-02): Update GPS data to include units, 🎬
Closes #73 (this PR supersedes that one)
Closes #77 (this PR supersedes that one too)
Closes #53\*
Closes #54†

This PR updates the representations of the GPS data to include units. Previously we were storing "unitless" numbers and specifying the types implicitly (e.g. comments and documentation) and now we are moving to `Quantity`s  with `Unit`s. This API (similar to #73 and #77, the first two attempts) is heavily inspired by [the JSR 363 specification](https://docs.google.com/document/d/12KhosAFriGCczBs6gwtJJDfg_QlANT92_lhxUWO2gCY) but I've tried my darned [... cut]
closed 2017-04-23 by whymarrh