LakeMaps/boat #56: Benchmark serial communication

whymarrh, 2017-03-21:
Refs #47

The serial communication needs to be benchmarked—I would like to know if we achieve 115200 bps.

arandell93, 2017-03-21:
As the microcontrollers are specified to communicate at 115200, if the boat or shoreside were not reaching this speed, the microcontrollers would not understand the messages.

whymarrh, 2017-03-26:
Yeah, that description doesn't make much sense. The idea behind this issue was to benchmark the serial communication in an attempt see whether or not 115200bps can be transferred from one end to the other (which, thinking about it now, is unlikely).

whymarrh, 2017-03-26:
The reason behind this: if possible I'd like to abstract over serial communication and where it'll obviously have a cost to it, I want to know if we can spare it.

whymarrh, 2017-04-02:
I'm moving this into the future milestone—I'll get to this when I have time to replace the serial lib.

whymarrh, 2017-04-08:
I'm closing this for now, I'll run benchmarks when I actually come up with a serial library replacement.

closed 2017-04-08 by whymarrh