LakeMaps/boat #67: Some NMEA sentences from the GPS are not parsed correctly

whymarrh, 2017-04-15:
Testing discovered that the NMEA sentences outputted from the GPS aren't always parsed correctly. It seems that some fields that were assumed to be required are not always provided and that sentences can oft be malformed.

A few examples:

```
$GPRMC,193440.398,A,47269,W,0.38,106.55,140417,,,A*3E
$GPGGA,1900,,,,,0,00,,,M,0.0,M,,000011,232,,18,03,082,,10,02,078,,20,01,024,*4A
$GPGSA,A,3,27,26,16,21,,,,,,,,,8.5,5.7,6.3*3D
$GPGSV,3,3,11,06,29,179,,20,10,277,,33,24,134,26*4E
$GPGSV,3,1,12,22,73,028,,15,73,036,,10,66,051,,26,62,095,43*7D
$GPGGA,21.000,V,,,,,,,140417,,,N*54
$GPRMC,013519.476,V,00,,,M,0.0,M,,0000*27
$GPGGA,0139GPGGA,013935.482,,,,,0,00,,,M,0.0,M,,0000*24
$GPRMC,014305.476,V,,0,,00,0.0,M,,0000*56
$GPGGA,53*7C
$GPRMC,015149.000,A,4735.1825,N,05244.1716,W,0.11,69.5717,03,,,,,,,,6.7,4.3,5.1*00
$GPRMC,020645.000,A,4735.1791,N,05244.1718,W,0.07,66.1*3D
$GPRMC,020825.000,A,4735.1835,N,05244,19,22,12,17,03,06,,,,,,,6.3,4.1,4.8*6E
$GPRMC,021250.000,A,4735.1803,N,05243,19,22,12,17,03,06,,,,,,,6.0,3.9,4.6*67
```

whymarrh, 2017-04-15:
All of the above messages seem to have **correct** checksums:

```
>>> checksum = lambda s: hex(reduce(lambda acc, c: acc ^ c, map(ord, list(s)), 0))
>>> checksum("GPRMC,193440.398,A,47269,W,0.38,106.55,140417,,,A")
'0x3e'
>>> checksum("GPGGA,1900,,,,,0,00,,,M,0.0,M,,000011,232,,18,03,082,,10,02,078,,20,01,024,")
'0x4a'
>>> checksum("GPGSA,A,3,27,26,16,21,,,,,,,,,8.5,5.7,6.3")
'0x3d'
>>> checksum("GPGSV,3,3,11,06,29,179,,20,10,277,,33,24,134,26")
'0x4e'
>>> checksum("GPGSV,3,1,12,22,73,028,,15,73,036,,10,66,051,,26,62,095,43")
'0x7d'
>>> checksum("GPGGA,21.000,V,,,,,,,140417,,,N")
'0x54'
>>> checksum("GPRMC,013519.476,V,00,,,M,0.0,M,,0000")
'0x27'
>>> checksum("GPGGA,0139GPGGA,013935.482,,,,,0,00,,,M,0.0,M,,0000")
'0x24'
>>> checksum("GPRMC,014305.476,V,,0,,00,0.0,M,,0000")
'0x56'
>>> checksum("GPGGA,53")
'0x7c'
>>> checksum("GPRMC,015149.000,A,4735.1825,N,05244.1716,W,0.11,69.5717,03,,,,,,,,6.7,4.3,5.1")
'0x0'
>>> checksum("GPRMC,020645.000,A,4735.1791,N,05244.1718,W,0.07,66.1")
'0x3d'
>>> checksum("GPRMC,020825.000,A,4735.1835,N,05244,19,22,12,17,03,06,,,,,,,6.3,4.1,4.8")
'0x6e'
>>> checksum("GPRMC,021250.000,A,4735.1803,N,05243,19,22,12,17,03,06,,,,,,,6.0,3.9,4.6")
'0x67'
```

whymarrh, 2017-04-15:
The most perplexing example is easily:

```
$GPGGA,0139GPGGA,013935.482,,,,,0,00,,,M,0.0,M,,0000*24
```

What am I missing here? Is this a valid message?

arandell93, 2017-04-15:
> What am I missing here? Is this a valid message?

It certainly appears to be invalid at first glance since the message is interrupted partway through. However, it appears the second 'attempt' is valid at a glance.

whymarrh, 2017-04-15:
The issue is that there's no way to sanely parse such an absurd message.

arandell93, 2017-04-16:
The only time you'll ever see a `GP` in succession is at the start of a message, and _always_ at the start of a message. If you receive a `GP`, start a new message and end it with the next checksum. Does that make any sense?

whymarrh, 2017-04-16:
I'll have to think about that—it strikes me as an assumption that shouldn't need to be made.

whymarrh, 2017-04-16:
I've created a gist with all of the sentences I have from yesterdays's testing: [gist:db6c0c88](https://gist.github.com/whymarrh/db6c0c88342576488cf31a8d14b01fcb)

arandell93, 2017-04-17:
So we only encountered a GP that was not preceeded by a $ once in ~3500 messages?
I'd say just call it invalid and skip it. Always look for a $GP to start a message and a *checksum to end, and then validate the checksum.

whymarrh, 2017-04-17:
> I'd say just call it invalid and skip it.

Agreed.

> Always look for a $GP to start a message and a *checksum to end, and then validate the checksum.

Kinda going back to my previous comment, that's not really what the spec<sup>\[1\]</sup> says. Sentences start with `$` and I don't really want to write a parser that does anything other than that. 😞

<sup>1\. In as much as there is a spec for NMEA sentences.</sup>

arandell93, 2017-04-17:
So was the issue that the sentence in question passed the checksum? I don't understand how it could have.

whymarrh, 2017-04-17:
It's odd, because (as one would expect) the checksum indeed would be different without the extra characters:

```
>>> checksum("GPGGA,0139GPGGA,013935.482,,,,,0,00,,,M,0.0,M,,0000")
'0x24'
>>> checksum("GPGGA,013935.482,,,,,0,00,,,M,0.0,M,,0000")
'0x55'
```

arandell93, 2017-04-17:
Can you specify that the checksum 'input' starts at the `$` and ends at the `*`?

whymarrh, 2017-04-17:
Hmm? The checksum-ed (is that a word?) bit is everything between the `$` and the `*` (exclusive).

arandell93, 2017-04-17:
Did the message in question pass or fail the checksum?

whymarrh, 2017-04-18:
Ah, sorry, the sentence in the OP did pass the checksum—all the sentences in the OP have correct checksums (where the checksum is calculated from the characters between the `$` and the `*`, exclusive). In [my comment above](https://github.com/LakeMaps/boat/issues/67#issuecomment-294624431) I further noting the peculiarities of the one sentence in question.

arandell93, 2017-04-18:
So what we are seeing is that the GPS actually appended a checksum that was calculated as described between `$` and `*` and we in turn verified that under those rules, the checksum was valid.

I'm not sure if we can easily detect the case where the GPS creates an invalid sentence with a valid checksum.

arandell93, 2017-04-30:
Some examples from the new GPS module at 57600 and 10Hz update:

```
$GPRMC,174825.000,V,,,,,0.00,0.00,300417,,,N*41
$GPVTG,0.00,T,,M,0.00,N,0.00,K,N*32
$GPGGA,174825.100,,,,,0,00,,,M,,M,,*74
$GPGSA,A,1,,,,,,,,,,,,,,,*1E
$GPRMC,174825.100,V,,,,,0.00,0.00,300417,,,N*40
$GPVTG,0.00,T,,M,0.00,N,0.00,K,N*32
$GPGGA,174825.200,,,,,0,00,,,M,,M,,*77
$GPGSA,A,1,,,,,,,,,,,,,,,*1E
$GPRMC,174825.200,V,,,,,0.00,0.00,300417,,,N*43
$GPVTG,0.00,T,,M,0.00,N,0.00,K,N*32
$GPGGA,174825.300,,,,,0,00,,,M,,M,,*76
$GPGSA,A,1,,,,,,,,,,,,,,,*1E
$GPRMC,174825.300,V,,,,,0.00,0.00,300417,,,N*42
$GPVTG,0.00,T,,M,0.00,N,0.00,K,N*32
$GPGGA,174825.400,,,,,0,00,,,M,,M,,*71
$GPGSA,A,1,,,,,,,,,,,,,,,*1E
$GPGSV,2,1,08,27,73,192,,26,42,081,,08,37,217,,09,31,280,*7B
$GPGSV,2,2,08,23,30,240,,21,29,064,,07,24,310,,31,07,138,*73
$GPRMC,174825.400,V,,,,,0.00,0.00,300417,,,N*45
$GPVTG,0.00,T,,M,0.00,N,0.00,K,N*32
$GPGGA,174825.500,,,,,0,00,,,M,,M,,*70
$GPGSA,A,1,,,,,,,,,,,,,,,*1E
$GPRMC,174825.500,V,,,,,0.00,0.00,300417,,,N*44
$GPVTG,0.00,T,,M,0.00,N,0.00,K,N*32
$GPGGA,174825.600,,,,,0,00,,,M,,M,,*73
$GPGSA,A,1,,,,,,,,,,,,,,,*1E
$GPRMC,174825.600,V,,,,,0.00,0.00,300417,,,N*47
$GPVTG,0.00,T,,M,0.00,N,0.00,K,N*32
$GPGGA,174825.700,,,,,0,00,,,M,,M,,*72
$GPGSA,A,1,,,,,,,,,,,,,,,*1E
$GPRMC,174825.700,V,,,,,0.00,0.00,300417,,,N*46
$GPVTG,0.00,T,,M,0.00,N,0.00,K,N*32
$GPGGA,174825.800,,,,,0,00,,,M,,M,,*7D
$GPGSA,A,1,,,,,,,,,,,,,,,*1E
$GPRMC,174825.800,V,,,,,0.00,0.00,300417,,,N*49
$GPVTG,0.00,T,,M,0.00,N,0.00,K,N*32
$GPGGA,174825.900,,,,,0,00,,,M,,M,,*7C
$GPGSA,A,1,,,,,,,,,,,,,,,*1E
$GPGSV,2,1,08,27,73,192,,26,42,081,,08,37,217,,09,31,280,*7B
$GPGSV,2,2,08,23,30,240,,21,29,064,,07,24,310,,31,07,138,*73
$GPRMC,174825.900,V,,,,,0.00,0.00,300417,,,N*48
$GPVTG,0.00,T,,M,0.00,N,0.00,K,N*32
$GPGGA,174826.000,,,,,0,00,,,M,,M,,*76
$GPGSA,A,1,,,,,,,,,,,,,,,*1E
$GPRMC,174826.000,V,,,,,0.00,0.00,300417,,,N*42
$GPVTG,0.00,T,,M,0.00,N,0.00,K,N*32
$GPGGA,174826.100,,,,,0,00,,,M,,M,,*77
$GPGSA,A,1,,,,,,,,,,,,,,,*1E
$GPRMC,174826.100,V,,,,,0.00,0.00,300417,,,N*43
$GPVTG,0.00,T,,M,0.00,N,0.00,K,N*32
$GPGGA,174826.200,,,,,0,00,,,M,,M,,*74
$GPGSA,A,1,,,,,,,,,,,,,,,*1E
$GPRMC,174826.200,V,,,,,0.00,0.00,300417,,,N*40
$GPVTG,0.00,T,,M,0.00,N,0.00,K,N*32
$GPGGA,174826.300,,,,,0,00,,,M,,M,,*75
$GPGSA,A,1,,,,,,,,,,,,,,,*1E
$GPRMC,174826.300,V,,,,,0.00,0.00,300417,,,N*41
$GPVTG,0.00,T,,M,0.00,N,0.00,K,N*32
$GPGGA,174826.400,,,,,0,00,,,M,,M,,*72
$GPGSA,A,1,,,,,,,,,,,,,,,*1E
$GPGSV,2,1,08,27,73,192,,26,42,081,,08,37,217,,09,31,280,*7B
$GPGSV,2,2,08,23,30,240,,21,29,064,,07,24,310,,31,07,138,*73
$GPRMC,174826.400,V,,,,,0.00,0.00,300417,,,N*46
$GPVTG,0.00,T,,M,0.00,N,0.00,K,N*32
$GPGGA,174826.500,,,,,0,00,,,M,,M,,*73
$GPGSA,A,1,,,,,,,,,,,,,,,*1E
$GPRMC,174826.500,V,,,,,0.00,0.00,300417,,,N*47
$GPVTG,0.00,T,,M,0.00,N,0.00,K,N*32
$GPGGA,174826.600,,,,,0,00,,,M,,M,,*70
$GPGSA,A,1,,,,,,,,,,,,,,,*1E
$GPRMC,174826.600,V,,,,,0.00,0.00,300417,,,N*44
$GPVTG,0.00,T,,M,0.00,N,0.00,K,N*32
$GPGGA,174826.700,,,,,0,00,,,M,,M,,*71
$GPGSA,A,1,,,,,,,,,,,,,,,*1E
$GPRMC,174826.700,V,,,,,0.00,0.00,300417,,,N*45
$GPVTG,0.00,T,,M,0.00,N,0.00,K,N*32
$GPGGA,174826.800,,,,,0,00,,,M,,M,,*7E
$GPGSA,A,1,,,,,,,,,,,,,,,*1E
$GPRMC,174826.800,V,,,,,0.00,0.00,300417,,,N*4A
$GPVTG,0.00,T,,M,0.00,N,0.00,K,N*32
$GPGGA,174826.900,,,,,0,00,,,M,,M,,*7F
$GPGSA,A,1,,,,,,,,,,,,,,,*1E
$GPGSV,2,1,08,27,73,192,,26,42,081,,08,37,217,,09,31,280,*7B
$GPGSV,2,2,08,23,30,240,,21,29,064,,07,24,310,,31,07,138,*73
$GPRMC,174826.900,V,,,,,0.00,0.00,300417,,,N*4B
$GPVTG,0.00,T,,M,0.00,N,0.00,K,N*32
$GPGGA,174827.000,,,,,0,00,,,M,,M,,*77
$GPGSA,A,1,,,,,,,,,,,,,,,*1E
$GPRMC,174827.000,V,,,,,0.00,0.00,300417,,,N*43
$GPVTG,0.00,T,,M,0.00,N,0.00,K,N*32
$GPGGA,174827.100,,,,,0,00,,,M,,M,,*76
$GPGSA,A,1,,,,,,,,,,,,,,,*1E
$GPRMC,174827.100,V,,,,,0.00,0.00,300417,,,N*42
$GPVTG,0.00,T,,M,0.00,N,0.00,K,N*32
$GPGGA,174827.200,,,,,0,00,,,M,,M,,*75
$GPGSA,A,1,,,,,,,,,,,,,,,*1E
$GPRMC,174827.200,V,,,,,0.00,0.00,300417,,,N*41
$GPVTG,0.00,T,,M,0.00,N,0.00,K,N*32
$GPGGA,174827.300,,,,,0,00,,,M,,M,,*74
$GPGSA,A,1,,,,,,,,,,,,,,,*1E
$GPRMC,174827.300,V,,,,,0.00,0.00,300417,,,N*40
$GPVTG,0.00,T,,M,0.00,N,0.00,K,N*32
$GPGGA,174827.400,,,,,0,00,,,M,,M,,*73
$GPGSA,A,1,,,,,,,,,,,,,,,*1E
$GPGSV,2,1,08,27,73,192,,26,42,081,,08,37,217,,09,31,280,*7B
$GPGSV,2,2,08,23,30,240,,21,29,064,,07,24,310,,31,07,138,*73
$GPRMC,174827.400,V,,,,,0.00,0.00,300417,,,N*47
$GPVTG,0.00,T,,M,0.00,N,0.00,K,N*32
$GPGGA,174827.500,,,,,0,00,,,M,,M,,*72
$GPGSA,A,1,,,,,,,,,,,,,,,*1E
$GPRMC,174827.500,V,,,,,0.00,0.00,300417,,,N*46
$GPVTG,0.00,T,,M,0.00,N,0.00,K,N*32
$GPGGA,174827.600,,,,,0,00,,,M,,M,,*71
$GPGSA,A,1,,,,,,,,,,,,,,,*1E
$GPRMC,174827.600,V,,,,,0.00,0.00,300417,,,N*45
$GPVTG,0.00,T,,M,0.00,N,0.00,K,N*32
$GPGGA,174827.700,,,,,0,00,,,M,,M,,*70
$GPGSA,A,1,,,,,,,,,,,,,,,*1E
$GPRMC,174827.700,V,,,,,0.00,0.00,300417,,,N*44
$GPVTG,0.00,T,,M,0.00,N,0.00,K,N*32
$GPGGA,174827.800,,,,,0,00,,,M,,M,,*7F
$GPGSA,A,1,,,,,,,,,,,,,,,*1E
$GPRMC,174827.800,V,,,,,0.00,0.00,300417,,,N*4B
$GPVTG,0.00,T,,M,0.00,N,0.00,K,N*32
$GPGGA,174827.900,,,,,0,00,,,M,,M,,*7E
$GPGSA,A,1,,,,,,,,,,,,,,,*1E
$GPGSV,2,1,08,27,73,192,,26,42,081,,08,37,217,,09,31,280,*7B
$GPGSV,2,2,08,23,30,240,,21,29,064,,07,24,310,,31,07,138,*73
$GPRMC,174827.900,V,,,,,0.00,0.00,300417,,,N*4A
$GPVTG,0.00,T,,M,0.00,N,0.00,K,N*32
$GPGGA,174828.000,,,,,0,00,,,M,,M,,*78
$GPGSA,A,1,,,,,,,,,,,,,,,*1E
```

arandell93, 2017-04-30:
[28000+ lines](https://gist.github.com/arandell93/f7c1bf798cc2067683aaae9e27c8a37c)

whymarrh, 2017-04-30:
[12853 from `sort < gistfile1.txt | uniq`](https://gist.github.com/whymarrh/17e476dd7306c0fdf272005cbd0a0b84)

whymarrh, 2018-03-10:
> I'm not sure if we can easily detect the case where the GPS creates an invalid sentence with a valid checksum.

Reading through this now, yup, I agree—I don't think there's a way to fix this. This is the world we'll have to live with. I imagine that the culprit here is that the checksum is calculated correctly on incorrect messages _after_ they're constructed and thus the checksum no longer serves as a way to verify the correctness of the sentence, leaving it to signify only that these are the bytes the GPS sent us. 🙈

closed 2018-03-10 by whymarrh