LakeMaps/boat #84: Router Custom Firmware Options

arandell93, 2017-05-10:
Based on information available on the internet, the WRT54G (V1.0, CS:CC99) router we are using is quite powerful and capable of performing quite well when configured correctly. Unfortunately, in order to get at these nifty settings we need to put custom firmware on it. Luckily, there is a lot of support for this.

The two main options for custom firmware are:

- [OpenWRT](https://en.wikipedia.org/wiki/OpenWrt) (Linux Terminal)

- [DD-WRT](https://en.wikipedia.org/wiki/DD-WRT) (GUI based)

We need to decide which is best suited for our purposes and then determine how to go about actually making the changes to the router.

Additionally, I have tried to upgrade the stock firmware on the WRT54G using the latest available [from Linksys](http://www.linksys.com/us/support-article?articleNum=148648) but it fails each time. Not sure what I'm doing wrong.

arandell93, 2017-05-27:
I have found that the router is not version 1 but is actually version 6.0. This means that is has less memory and generally poorer performing hardware than some other versions, but DD-WRT still has [a firmware build for it called micro](http://www.dd-wrt.com/wiki/index.php/Linksys_WRT54G_v5.0_%26_5.1_%26_6.0).

whymarrh, 2017-05-27:
So you're telling me there's a chance

arandell93, 2017-05-27:
To attempt it I need a hardwired connection and I don't have an ethernet adapter currently.

arandell93, 2017-05-27:
Current firmware:
[image]

On Linksys' website they don't list a firmware upgrade for this version, though I have found [pages discussing upgrades](http://community.linksys.com/t5/Wireless-Routers/Trying-to-upgrade-firmware-for-WRT54G-V6-on-Windows-7/m-p/293283#M153597).