LakeMaps/boat #90: Implement hardware serial GPS

whymarrh, 2017-06-01:
Refs #40

We will need to a new GPS with a higher update rate.

whymarrh, 2017-06-01:
@arandell93 would you mind linking the data sheet for the new GPS?

arandell93, 2017-06-01:
[Datasheet](https://cdn-shop.adafruit.com/datasheets/GlobalTop-FGPMMOPA6H-Datasheet-V0A.pdf)

arandell93, 2017-06-01:
For my own reference. We will need to use PI GPIO pins 4, 6, 8, 10 which are +5, GND, TX and RX respectively.
[image]

whymarrh, 2017-06-14:
Just to confirm that I'm not missing something here—we're simply switching between using the passthrough microcontroller to connecting the same GPS directly to the Pi, yeah? (I'm not sure why I previously referred to a "new" GPS.)

whymarrh, 2017-06-14:
As for the battery backup (from the link):

> Normally, if the GPS loses power, it will revert to the factory default for baud rates, configuration, etc. A backup battery will mean that those defaults will not be lost!
>
> The backup real-time-clock circuitry draws 7 uA (0.007 mA) so a CR1220 will last 40mAh / 0.007mA = 5,714 hours = 240 days continuously. The backup battery is only used when there's no main 3V power to the GPS, so as long as it's only used as backup once in a while, it will last years

No thanks—we will need to add the ability to configure the GPS at startup anyway so we may as well always *need* do it.

arandell93, 2017-06-15:
> we're simply switching between using the passthrough microcontroller to connecting the same GPS directly to the Pi, yeah? (I'm not sure why I previously referred to a "new" GPS.)

It is functionally the same, but we are in fact going from using the old GPS to using the new ones we recently bought. That's why we need to do the settings.

> No thanks—we will need to add the ability to configure the GPS at startup anyway (for when the battery inevitably dies) so we may as well always need do it.

I am okay with this.

whymarrh, 2017-06-16:
> It is functionally the same, but we are in fact going from using the old GPS to using the new ones we recently bought. That's why we need to do the settings.

That would make sense. It would also explain why I previously referred to a "new" GPS. The confusion stemmed from the fact that the linked datasheet is the same datasheet I wrote the current GPS implementation against (which is a good thing I guess). Anyhow, it all makes sense now.

arandell93, 2017-06-16:
_
           /(|
          (  :
         __\  \  _____
       (____)  `|
      (____)|   |
       (____).__|
        (___)__.|_____

PR #107 (2017-06-23): Implement hardware serial GPS
Closes #40
Closes #90

This PR updates the control software for the new GPS receiver. All the sentence parsing is the same.

At start-up we now:

1. Connect to the receiver using 9600 baud
1. Set the NMEA baud rate to 57600 (`PMTK_SET_NMEA_BAUDRATE`)
1. Disconnect and reconnect using 57600 baud
1. Set the NMEA update rate to 10 Hz (`PMTK_SET_NMEA_UPDATERATE`)
closed 2017-06-23 by whymarrh