LakeMaps/boat #135: Implement Gyro for more Accurate Navigation

arandell93, 2018-02-08:
To be completed following the completion and testing of #93. 

The use of the gyro will enable us to track our rotation rate (and approximate angular position) more accurately when used in combination with the magnetometer. There will be some math involved in creating what can effectively be described as a filter on our heading. 

Once we get #93 complete we can begin to look into this.

cgregory52, 2018-03-26:
In order to make a tightly coupled state estimator we'll need the gyro as well as the accelerometer and magnetometer. I personally feel this issue should still be part of #93.

whymarrh, 2018-03-27:
Good news, it'll all probably come in one package! (That is, it'll all be implemented at the same time.)