LakeMaps/boat #145: Date is incorrect on Pi

arandell93, 2018-02-20:
The Pi thinks every day is 2017-04-10, thus all log files appear to have been made then, despite the log files being named incorrectly per #144.

@FifoIronton has a screen shot for details.

arandell93, 2018-02-20:
Idea: Can we get it to pull real-world from either:

1. The shoreside (or any) computer once the network is up (ideal)
2. The GPS once the satellites get signal (not-ideal)

whymarrh, 2018-02-20:
The GPS would be ideal IMO, as that's the most accurate time source (atomic clocks and whatnot). We can certainly pull from the shoreside by default and use the GPS to adjust once we have signal.

arandell93, 2018-02-20:
Yeah let's do that. At least then we will always have a system time regardless of GPS state, but can use GPS for real pretty accuracy once we get it.

closed 2018-02-26 by whymarrh